STMicroelectronics/STM32CubeL4 #98: Update request Critical Security Vulnerabilities in FreeRTOS v10.3.1 (CVE-2021-32020, CVE-2021-31572.)

eduardoaugustojulio, 2025-01-15:
**Caution**

The Issues are strictly limited for the reporting of problems encountered with the software provided in this project.

For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools, or the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

  

---

  

**Describe the set-up**

-  **Board**: Custom board using the STM32L476.

-  **IDE/Compiler**: STM32CubeIDE version 1.17.0 with STM32Cube MCU Package for STM32L4 Series 1.18.1.

  

---

  

**Describe the bug**

The STM32Cube MCU Package for STM32L4 Series 1.18.1 uses **FreeRTOS v10.3.1**, which is affected by the following critical security vulnerabilities:

  

1.  **[CVE-2021-32020](https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2021-32020)**: Insufficient bounds checking during management of heap memory.

2.  **[CVE-2021-31572](https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2021-31572)**: Integer overflow in `stream_buffer.c` for a stream buffer.

3.  **[CVE-2021-31571](https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2021-31571)**: Integer overflow in `queue.c` for queue creation.

4.  **[CVE-2021-43997](https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2021-43997)**: Lack of prevention for non-kernel code from calling `xPortRaisePrivilege` to raise privilege.

  

These vulnerabilities pose significant risks, including unauthorized access, application instability, and denial of service.

  

---

  

**How To Reproduce**

  

1.  **Global Behavior**: The project demonstrates standard FreeRTOS-based multitasking behavior on STM32 boards.

2.  **Suspected Modules**: FreeRTOS kernel, `stream_buffer`, `queue`, `xPortRaisePrivilege` management, and memory handling.

3.  **Use Case**: Applications involving tasks with high-priority interrupts or extensive memory operations may trigger these vulnerabilities.

4.  **Reproduction Steps**:

- Set up a project using **FreeRTOS v10.3.1** on **STM32CubeIDE v1.17.0** with the **STM32Cube MCU Package for STM32L4 Series 1.18.1**.

  

---

  

**Additional context**

Looking at the STM32 Github repo [stm32-mw-freertos repository](https://github.com/STMicroelectronics/stm32-mw-freertos) **FreeRTOS v10.6.2** is already available. This version includes fixes for the mentioned vulnerabilities. I am happy to assist with testing or integration if needed.

  

---

  

**Screenshots**

Not applicable for this issue.

edsonms, 2025-01-15:
I have the same issue. Can't find a way to upgrade FreeRTOS to v10.6.2 on STM32CubeMx. Version v10.3.1 as explained on the ticket has cybersecurity vulnerabilities.

ALABSTM, 2025-01-17:
Hi @eduardoaugustojulio,

Thank you for having reported. We are already aware of this point and are working on it. By the way, please note such topics related to vulnerabilities shall be reported elsewhere, as explained in the [SECURITY.md](https://github.com/STMicroelectronics/STM32CubeL4/blob/master/SECURITY.md) file.

With regards,

ALABSTM, 2025-01-30:
ST Internal Reference: 190676

ALABSTM, 2025-01-30:
Hi @eduardoaugustojulio and @edsonms,

Our teams are working on upgrading to the new `FreeRTOS` version. This may take time. We cannot share a date for the moment. I will keep you informed.

With regards,