STMicroelectronics/STM32CubeL4 #99: Don't misuse __TIME__

CanastraRF, 2025-02-07:
From ANSI C there are few predefined macros

__DATE__, __FILE__, __LINE__, __STDC__, __STDC_VERSION__, __TIME__, __cplusplus

But You misuse __TIME__ in different macros.

IS_ADC_SAMPLE_TIME(__TIME__)
IS_LCD_DEAD_TIME(__TIME__)
IS_UART_ASSERTIONTIME(__TIME__)
IS_UART_DEASSERTIONTIME(__TIME__)
IS_FMC_ADDRESS_SETUP_TIME(__TIME__)
IS_FMC_ADDRESS_HOLD_TIME(__TIME__)
IS_FMC_DATASETUP_TIME(__TIME__)
IS_FMC_TURNAROUND_TIME(__TIME__)
IS_FMC_MAX_CHIP_SELECT_PULSE_TIME(__TIME__)
IS_FMC_TCLR_TIME(__TIME__)
IS_FMC_TAR_TIME(__TIME__)
IS_FMC_SETUP_TIME(__TIME__)
IS_FMC_WAIT_TIME(__TIME__)
IS_FMC_HOLD_TIME(__TIME__)
IS_FMC_HIZ_TIME(__TIME__)

Please replace __TIME__ in all macro with a not predefined macro.

BTW: all symbols beginning with _ are reserved for the compiler and his library 
and should not used in other libraries or user code

You can begin you symbols with HAL_ or STM_

Reto Felix

ASEHSTM, 2025-02-13:
Hello @CanastraRF,

Thank you for your report. I understand your point and will share it with our development team. Since this modification requires refactoring due to the widespread use of this macro. Thank you for your understanding.

With Regards,

CanastraRF, 2025-02-14:
A simple solution is to replace __ TIME __ with TIME__.
This is only necessary in the above macro.
All other __ symbols __ can replaces later.

ASEHSTM, 2025-03-10:
Hello @CanastraRF,

I understand your point. However, on our side, we haven't encountered any issues with this naming convention that would necessitate refactoring. 
Could you please specify if you have encountered any problems with these parameters, or if you are based on a visual review?

With Regards,

CanastraRF, 2025-03-11:
Hello ASEHSTM
It is only a visual review.

ASEHSTM, 2025-03-11:
Hello @CanastraRF,

Since we have not encountered any issues, we do not have any plans to replace these parameters. Therefore, please allow me to close this issue.

Thank you for your understanding.

With regards,

closed 2025-03-11 by ASEHSTM