STMicroelectronics/STM32CubeL4 #100: Don't misuse COMP1, COMP2

CanastraRF, 2025-02-14:
Dear Support

You define in lot of CPU files

#define COMP1               (COMP_TypeDef *) COMP1_BASE
#define COMP2               (COMP_TypeDef *) COMP1_BASE

But CMSIS also use COMP1 or COMP2 in  DWT_Type in core_cm4.h

So depending of the include order of above file one of both symbols is not accessible or result in compile errors.

Please don't use preprocessor for most definition

Better use 
static COMP_TypeDef * const COMP1 = (COMP_TypeDef *) COMP1_BASE;
static COMP_TypeDef * const COMP2 = (COMP_TypeDef *) COMP2_BASE;

Modern Compiler can remove unused const
Reto Felix

TOUNSTM, 2025-04-24:
Hello @CanastraRF,

Thank you for your report. I tried to reproduce the issue on my side but did not encounter any problems. Could you please provide more details on how we can reproduce this issue?

Best regards,

CanastraRF, 2025-04-28:
Hi TOUNSTM
I create a small demo project for Keil MDK.
It is not so difficult 
Add 
`  DWT_Type	dwt_Type; `
`  dwt_Type.COMP0 = 12345678UL; `
`  dwt_Type.COMP1 = 12345678UL; `
`  dwt_Type.COMP2 = 12345678UL; `
`  dwt_Type.COMP3 = 12345678UL; `

[CompTest.zip](https://github.com/user-attachments/files/19937691/CompTest.zip)

to a source file where COMP1 or COMP2 from a stm32l431xx.h is included
Best regards

TOUNSTM, 2025-05-26:
Hello @CanastraRF,

I hope you are doing well. We finally got the feedback of our development teams and you are absolutely right about the conflict between naming in `stm32l431xx.h` and `core_cm4.h`. Unfortunately, it's not possible to change CMSIS device header file nor ARM core_cm4.h file.  The solution is to isolate the code that uses DWT into separate file(s) that only include the `core_cm4.h` file and not the ST device one

I hope this clarifies the situation. Please allow me to close this issue.

Best regards,

closed 2025-05-26 by TOUNSTM